AllLuckZeroSkill/ECHO #22: Evaluate Applicability of Compliance Standards for Spatial Awareness Device

AllLuckZeroSkill, 2024-03-06:
### Objective: 
To explore and ascertain the relevance of various compliance standards to the development of the spatial awareness device, ensuring alignment with safety, health, and environmental regulations.

### Task Overview: 
Our project involves the integration of audio processing, directional feedback via vibration motors, and Bluetooth connectivity for enhancing user interaction. This task involves researching potential standards that could guide the design and development process, ensuring the product's safety, efficacy, and market readiness. It's important to note that not all listed standards may be directly applicable to our project; the goal is to identify those that are pertinent.

### Deliverables:

A list of investigated standards, including but not limited to:

- IEC/UL 62368-1: Safety requirements for audio/video, information, and communication technology equipment.
- EN 50566: 2013 / IEC 62209-2: Assessment of exposure to electromagnetic fields for wearable devices.
- IEC 60601-1: General requirements for basic safety and essential performance of medical electrical equipment.
- Bluetooth SIG Qualification: Compliance with Bluetooth Special Interest Group standards for wireless communication.
- IP Rating: Examination of Ingress Protection rating requirements for durability and resistance to environmental elements.
- An analysis detailing how each standard could potentially apply to our device, focusing on user interaction, wireless communication, and physical design aspects.
- Recommendations for aligning our design and testing protocols with these standards, if applicable.

### **Deadline: Monday March 11th**

MarkWasz, 2024-03-08:
Brief overview of each is: 
CFR - Code of Federal Regulations Title 21 (21CFR820.70) of the FDA for Medical Devices provides some good general standards that could apply to our project. These are things that we are already doing for our project, so it would be nice to say that we are taking inspiration from the following standards:
(1) Documented instructions, standard operating procedures (SOP's), and methods that define and control the manner of production;
(3) Compliance with specified reference standards or codes; 
(5) Criteria for workmanship which shall be expressed in documented standards or by means of identified and approved representative samples.
(b) Production and process changes. Each manufacturer shall establish and maintain procedures for changes to a specification, method, process, or procedure. Such changes shall be verified or where appropriate validated according to § 820.75, before implementation and these activities shall be documented. Changes shall be approved in accordance with § 820.40.
(c) Environmental control. Where environmental conditions could reasonably be expected to have an adverse effect on product quality, the manufacturer shall establish and maintain procedures to adequately control these environmental conditions. Environmental control system(s) shall be periodically inspected to verify that the system, including necessary equipment, is adequate and functioning properly. These activities shall be documented and reviewed.
(d) Personnel. Each manufacturer shall establish and maintain requirements for the health, cleanliness, personal practices, and clothing of personnel if contact between such personnel and product or environment could reasonably be expected to have an adverse effect on product quality. The manufacturer shall ensure that maintenance and other personnel who are required to work temporarily under special environmental conditions are appropriately trained or supervised by a trained individual.
 
https://www.accessdata.fda.gov/scripts/cdrh/cfdocs/cfcfr/CFRSearch.cfm?fr=820.70

EN 50566:2013 “Product standard to demonstrate compliance of radio frequency fields from handheld and body-mounted wireless communication devices used by the general public (30 MHz - 6 GHz)" 
- Mounted on the body.
- Combined with other transmitting or non-transmitting devices or accessories (such as belt-clips, cameras, or Bluetooth add-ons).

https://standards.globalspec.com/std/14626315/en-50566

IEC 62209-2 is titled “Human exposure to RF fields from hand-held and body-mounted wireless communication devices – Human models, instrumentation, and procedures – Procedure to determine the specific absorption rate (SAR) for wireless communication devices used in close proximity to the human body,” provides guidelines for assessing exposure to electromagnetic fields. This applies somewhat to our project, but EN 50566:2013  is more relevant. 

IEC 60601-1 called Medical electrical equipment – Part 1: General requirements for basic safety and essential performance.
It provides general safety requirements applicable to various medical devices. It ensures that these devices meet safety criteria related to electrical shock, mechanical hazards, radiation, and other risks. The standard is widely accepted as a benchmark for medical equipment safety. Note that this standard does not guarantee the effectiveness of a medical device; effectiveness evidence is typically required by regulatory bodies like the FDA in the US. 
https://en.wikipedia.org/wiki/IEC_60601

Bluetooth SIG Qualification is for The Bluetooth Qualification Process reinforces the strength of the Bluetooth brand and ecosystem. It benefits all Bluetooth SIG members by promoting product interoperability.
Qualification ensures that Bluetooth products comply with the Bluetooth Patent & Copyright License Agreement and the Bluetooth Trademark License Agreement (collectively, the Bluetooth License Agreement) as well as the Bluetooth Specifications.
https://www.bluetooth.com/develop-with-bluetooth/qualification-listing/

An IP Rating is for Ingress Protection and is used to specify the extent of protection an electrical enclosure provides against environmental conditions. These ratings are crucial for ensuring the safety and reliability of various electronic devices. Let’s explore the details:

MarkWasz, 2024-03-08:
Analysis on each:
CFR - Code of Federal Regulations Title 21 (21CFR820.70) of the FDA  is generic and is applies to the testing/practices that we already do. 
https://www.accessdata.fda.gov/scripts/cdrh/cfdocs/cfcfr/CFRSearch.cfm?fr=820.70

The EN 50566 and IEC 62209-2 standards significantly impact your wireless communication device in terms of safety, compliance, and user well-being. The close proximity of our system will apply to these standards.  Since our system is used within 200 mm from the human body, this standard becomes particularly relevant.
Risk mitigation of the system is also relevant here also. By adhering to EN 50566, we can minimize potential health risks associated with EMF exposure. Although not something we are directly testing, it would show that we are aware of what it is and how to prevent that.

IEC 60601-1 goes over the same stuff as the FDA 21CFR820.70 except that 21CFR820.70 is more general and easier to apply to our project. 

IP Rating is more for testing the intended environment for your system (e.g., indoor, outdoor, industrial) and to identify the level of protection needed (e.g., minimal dust exposure, water resistance). This could apply to our "bump and drop" testing.

closed 2024-03-11 by MarkWasz